LumaTeam/Luma3DS #1458: Godmode9 1.9.1 does not work with Luma3DS 10.2

sthetix, 2020-07-17:
I found that Godmode9 1.9.1 does not work with Luma3DS 10.2, any info on it?

Thanks

Oreo639, 2020-07-17:
Duplicate:
https://github.com/LumaTeam/Luma3DS/issues/1457#issuecomment-659756949

Shadow322, 2020-07-17:
> I found that Godmode9 1.9.1 does not work with Luma3DS 10.2, any info on it?
> 
> Thanks

Works here

TheMasterOfMike, 2020-07-17:
If you have two payloads, it will work fine (chainloader menu loads up) but if it's only one, and you're using B9S and Luma 10.2, it will not work.

Shadow322, 2020-07-17:
> If you have two payloads, it will work fine (chainloader menu loads up) but if it's only one, and you're using B9S and Luma 10.2, it will not work.
> 
> That being said, it's already been reasonably proven it's an issue with GodMode9.

Ah I see I do have 2 payloads then

closed 2020-07-17 by TuxSH