LumaTeam/Luma3DS #1462: Luma3d not patching games

LucarioRanger, 2020-07-18:
I have been trying to create an Omega Ruby Randomize. My issue is that luma3d does not patch the game at all. Everything is in the SD card in the right place, but nothing patches and it plays like normal. I have the config set to 'enable game patching' but still nothing.
I have also had no luck on the home brew discord

I have no update data installed
I fallowed this guide https://zetadesigns.github.io/randomizing-layeredfs.html
My files are in my SD card as luma/titles/titleID/(All modified files)

Any advice? Also I am on Luma v10. 1.3 if that matters and the game ROM was dumped from a cartridge

TuxSH, 2020-07-18:
The feature is working fine on Alpha Sapphire... is Luma3DS launched from the SD card? (bring up the luma config menu, then there's text in yellow in the bottom screen, bottom left corner)

acedarth, 2020-07-21:
im having the same issue but the text in the bottom left is blue not yellow

Juancarlo577, 2021-01-23:
Make sure the titleid matches with the gameid via fbi titles and not just titleid is mostly my guess if you just named the folder titleid :/

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai