LumaTeam/Luma3DS #1467: Pokemon Ultra Sun/Moon can't use 'quick connect' after update Luma3DS to v10.2

swufexym, 2020-07-24:
System model:
**Old 2DS**

SysNAND version (+emu/redNAND version if applicable):
**11.13.0-45U SysNAND**

Entrypoint (How/what you're using to boot Luma3DS):
**Boot9Strap**

Luma3DS version:
**v10.2 stable**

Steps to reproduce:
**Update Luma3DS version to v10.2 and use 'Quick Connect' to connect with another old 2DS locally in the listed games. This function works in v10.1.3. I tried reconnect my 2DS to the same wifi and restart my 2DS but failed.**

Ammako, 2020-07-26:
You haven't properly filled in the bug report form

Can you confirm that Set developer UNITINFO is not enabled?

swufexym, 2020-08-01:
> You haven't properly filled in the bug report form
> 
> Can you confirm that Set developer UNITINFO is not enabled?

Thank you for reminding me that! The function acts appropriately after I disabled 'developer UNITINFO' selection.

fox8091, 2020-08-12:
> Thank you for reminding me that! The function acts appropriately after I disabled 'developer UNITINFO' selection.

Would you please close the issue then? Developer UNITINFO breaks things by design, as it makes your 3DS appear to be a developer console, and by proxy means that you can't use Nintendo online services. So there's no real issue?