LumaTeam/Luma3DS #1468: Most ram cheats only get poked

dariendark, 2020-07-25:
I have been testing many cheats taken directly from checkpoint and i have noticed that the ram/pointer cheats only poke the value. Hence you have to reopen the cheat menu over and over to turn them off and back on to repoke it as oppose to a value freeze. Also ive noticed the rom value cheats are irreversable once applied. Usually requiring a restart of the game to be cleared. Hence turning the cheat off doesn't poke them back. Is this normal or am I doing something wrong?
I've already tried google. Couldn't find anything.

dariendark, 2020-07-25:
Nvm. Figured it out. For anybody else who runs across the same issue. I just needed to update my version of luma.

Shadow322, 2020-07-25:
then close this issue

closed 2020-07-25 by dariendark