LumaTeam/Luma3DS #1470: GodMode9 don't boot after luma3ds update

davidcc73, 2020-07-27:
-- Please fill in the placeholders.-->
**System model:**

[New 2DS XL, **New 3DS XL**, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. **Boot9Strap**/Sighax, etc.]

**Luma3DS version:**

[e.g. **v10.2** stable or if using non-releases specify the commit like this 

**Steps to reproduce:**
After I update my luma3ds to the version v10.2 the godmode9 stopped working when I press the start during the boot, the 3ds only shows a black screen, I have to hold the power button to make it shut down. 
I tried to use de version v1.7.1 of godmod9 and it worked.

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: ( before)

Splash duration: ( 3s)

PIN lock: (off )

New 3DS CPU: (off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x )

Show NAND or user string in System Settings: ( x)

Show GBA boot screen in patched AGB_FIRM: ( x)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

toxic9, 2020-07-29:
Hi!

This is a known (godmode9) bug. Not Luma's.
The workaround is to have more than one payload on the same folder.
You can even copy godmode9 payload and call it godmode9(1).firm !!

Lesking72, 2020-07-29:
Another workaround for this issue that works for me is to use a splash and set Luma3DS to display the splash before payloads

Shadow322, 2020-07-29:
Just compile latest source as it's been fixed

toxic9, 2020-07-30:
> Just compile latest source as it's been fixed

Compile Luma3DS or godmode9 ??

Shadow322, 2020-07-30:
> > Just compile latest source as it's been fixed
> 
> Compile Luma3DS or godmode9 ??

It's a godmode9 side problem

closed 2020-08-04 by TuxSH

mackal, 2020-08-04:
Personally I copied both dev and normal firmware over, doing this resulted it bringing up the boot select menu, which has worked 100% of the time booting into GM9 after selecting it.

Shadow322, 2020-08-04:
> Personally I copied both dev and normal firmware over, doing this resulted it bringing up the boot select menu, which has worked 100% of the time booting into GM9 after selecting it.

Duh having more then one payload is a temp fix but just compile latest source as it's been fixed