LumaTeam/Luma3DS #1471: Takes a long time to power off from Test Menu

Ammako, 2020-07-28:
**System model:**

Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.12U

**Entrypoint (How/what you're using to boot Luma3DS):**

fastboot3ds

**Luma3DS version:**

10.2

**Luma3DS configuration/options:**

Screen brightness: (1)

Splash: (Off)

Splash duration: (1s)

PIN lock: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: (x)

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**

Powering off from Test Menu takes quite a while before the system finishes powering off. It works fine from Home Menu (regardless of unitinfo) and works fine if powering off from Rosalina. On 10.1.3 it was fine.

**Steps to reproduce:**

1. Boot into Test Menu

2. Power off

**Dump file:**

No dump file as it doesn't produce errors.

TuxSH, 2020-08-04:
How long? Not that I care of testmenu anyway.

Ammako, 2020-08-04:
Like 15 seconds

Ammako, 2020-08-05:
Breaks from 22db344

closed 2022-06-05 by suppai