LumaTeam/Luma3DS #1475: [Suggestion] Saving N3DS settings to a specific game

CatsLover2006, 2020-07-30:
There it is, in the title.
This is primarily so I can play the Pokémon games with a high frame rate while also being able to play Sonic & All Stars Racing Transformed, which crashes with L2 cache or a higher clock rate upon transforming (data steaming _too_ fast?)

Nuck-TH, 2020-07-30:
Actually, just blacklist would be better since only few games(4?) have issues with N3DS clockspeed/L2 cache.

IceCloud76, 2020-08-02:
I was searching the wiki for this because I thought that this would have been already been implemented, similar to how Twilight Menu ++ does with DS games. I would love to see this implemented too

TuxSH, 2020-08-04:
Use the Rosalina menu option instead of the luma main menu one.

closed 2020-08-04 by TuxSH

IceCloud76, 2020-08-04:
I think this feature still makes sense. Leaving the higher cpu clock and L2 cache enabled by default like the wiki recommends leads to some games crashing. Going back to turn off this AFTER the game crashes is cumbersome.