LumaTeam/Luma3DS #1476: Battery level indicator is always showing full charge and LED always stays blue.

Zafictus, 2020-08-03:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

You cant really see the actual power level on the main 3ds menu, although Rosalina and GM9 seem to be able to read battery level accurately.
Using version 11.13 europe.

**Steps to reproduce:**

1. Just installed the Luma 3ds, Rosalina and GM9 using the 3ds hacks guide with seedminer.

2.

**Dump file:**

TuxSH, 2020-08-04:
That's normal it's just how it is. See https://www.3dbrew.org/wiki/PTM:GetBatteryLevel for the battery levels.

closed 2020-08-04 by TuxSH

Zafictus, 2020-08-05:
So if you hack the 3ds you lose the battery indicator's red light warning for low battery? That makes zero sense.

profi200, 2020-08-05:
No, that's working perfectly fine with and without Luma. The battery indicator is handled by the MCU. Either there is something wrong with the MCU (can be corrupted firmware but Luma doesn't touch it at all) or the fuel gauge has failed and reports the wrong voltage.

Zafictus, 2020-08-08:
thank you. I ll format the 3ds and redo the procedure.