LumaTeam/Luma3DS #1481: NTP client failing

ThatNerdyPikachu, 2020-08-07:
**System model:**

N2DS

**SysNAND version (+emu/redNAND version if applicable):**

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap

**Luma3DS version:**

10.2 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Using the NTP client either results in freezing, or `Operation failed. (ffffffff)`

`time.windows.com` is not blocked, and is reachable.

**Steps to reproduce:**

1. Use NTP Client to set to -0500
2. Observe freezing or error message

ThatNerdyPikachu, 2020-08-07:
It started working all of a sudden? Strange.

closed 2020-11-17 by AuroraWright

TCAtrevor, 2023-09-13:
This is happening to me. The time sync menu completely freezes until I cut the network connection, at which point it returns the `Operation failed (ffffffff).` message.

therealprocyon, 2023-11-04:
For me it currently is occurring. It actually worked in the past but now it freezes and then gives this error.

salt86, 2024-11-08:
I was having a similar issue. My wireless was switched off. I turned it on and it worked.