LumaTeam/Luma3DS #1483: Opening the Rosalina Menu for 7+ seconds causes the Phoenix Wright Trilogy to hang

iomintz, 2020-08-08:
## Configuration

**System model:** New 3DS XL

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):** 11.13.0-45U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** v10.2 stable

**Luma3DS configuration/options:**

Default EmuNAND: N/A

Screen brightness: 4

Splash: Before payloads

Splash duration: 1 second

PIN lock: Off

New 3DS CPU: Off

--

Autoboot EmuNAND: N/A

Use EmuNAND FIRM if booting with R: N/A

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

## Explanation of the issue

On a cold boot, opening the Rosalina menu for more than a few seconds causes the Phoenix Wright Trilogy game to hang.

### Steps to reproduce

1. Launch the Phoenix Wright: Ace Attorney Trilogy.
2. Open the Rosalina menu for 7 seconds or more. If this is a cold boot, taking a screenshot will also do the trick.
3. Close the Rosalina menu.

### Expected behavior

The game is usable.

### Actual behavior

The game hangs. No buttons can be pressed, dialog text does not advance, and the music (if any) keeps playing. The home button must be pressed several times before the home menu opens.

iomintz, 2020-08-26:
Can confirm for the EarthBound Virtual Console release as well.

CebolaBros64, 2020-11-04:
A similar (if not the same) issue has been happening to me in Tomodachi Life. I've only been able to reproduce it with the 3D slider turned on.
Everything seems to take longer to load, but the game itself technically isn't "frozen". In some occasions, you can still interact with the touch screen. The music still plays, as well as the character voices.

youngmaydoria, 2022-02-25:
Can also confirm in ace attourney trilogy

lkchx123, 2022-05-02:
Can also confirm in Gyakuten Saiban 123 and M&S at the Rio2016 Olympic Games. I found this problem after changing the L2 + clock setting. But I revert back to the original settings, the problem still exists. So I'm not sure.

rocca91, 2022-06-12:
Can confirm it happens in fire emblem fates, I also hard reset in game and I got a long black screen causing me to force my 3ds to shut down, on the latest luma 11.0  didn't fix anything...

aripjatk, 2022-10-18:
I'm having the exact same issue with Ace Attorney Trilogy on my (Old) 3DS XL with the newest Luma version and firmware 11.16, regardless of whether the 3D slider is on or off.

cooolgamer, 2022-10-26:
Happened in project Mirai DX, the game is still running in background, it just show 2 frames but like I said, run fine in background but is incredibely slow, this is why you think it froze but it didn't, noticing this made me able to save the game in that situation

portasux, 2022-12-04:
unable to replicate with luma v11.0 and system version 11.16, i tried it with AC:NL WA (with vapecord, cartinstall) and NSMB2 (with the super mario bros next: the lost levels part 2 and a mod that replaces mario and luigi with their sm64 versions, came with console bundle)

Enderspearl184, 2022-12-26:
This just happened on my n2dsxl in Rhythm Heaven Megamix. (downloaded from the eshop) The top screen flickered between two frames of the pause menu being barely open but otherwise didn’t change, so this bug is still an issue.

unlikely-charlie, 2023-02-06:
Similar bug on Apollo Justice: Ace Attorney.

Nemris, 2023-02-06:
I've had it happen on a New3DS running Super Smash Bros. from an EUR cart, on latest Luma and firmware.

In my case, New3DS CPU is set to `Clock+L2`, but I can provide the full configuration and help testing commits if needed.

Omegoat, 2024-01-30:
Same with Team Kirby Clash Deluxe

bneils, 2026-08-20:
https://github.com/LumaTeam/Luma3DS/issues/2099#issuecomment-2525262503

> due to implementation limitation, can't fix.

Any way of getting clarification on this statement?