LumaTeam/Luma3DS #1486: [Rosalinda][Feature request] Screenshots

T0biasCZe, 2020-08-17:
Add config options to rosalina screenshot function, it could set which screen take (bottom/top/both), and if 3D is enabled, if both left/right screen is saved.
It could save a lot of waiting for screenshots to save.
Even the latest update should have it enhanced, it still takes 12seconds (new 3ds) to write the screenshot screenshot with 3D enabled

Shadow322, 2020-08-17:
Only takes 11ms converting and writing files 231ms 3d on full new 3ds xl

T0biasCZe, 2020-08-17:
I tested it 8times, and it converted shortly for less than 3times
Otherwise always 6+ seconds

Shadow322, 2020-08-17:
> I tested it 8times, and it converted shortly for less than 3times
> Otherwise always 6+ seconds

Bad SD card

AuroraWright, 2020-11-17:
Screenshots are really quick now on my console, so you likely have a slow SD card

closed 2020-11-17 by AuroraWright

iomintz, 2020-11-17:
@AuroraWright 

```
➤ dd if=/dev/zero of=$HOME/mnt/ThreeDS/1GB oflag=nocache conv=notrunc,fdatasync bs=1M count=1024
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 25.5708 s, 42.0 MB/s
```

Given that a screenshot of three screens takes up 806,562 bytes, this cannot be the cause.

iomintz, 2020-11-17:
Hm… I did another test on a New 3DS XL with 3D on, it takes me 4.3s, and with 3D off, 3.6s. @AuroraWright, @T0biasCZe are those the speeds you get? I timed it with a stopwatch rather than the built in timer since the built in one seems to vastly under-report.

T0biasCZe, 2020-11-17:
i tried it again, and with 3D on with external stopwatch i get 4,999s, built in 215ms.
with 3D off with ext. sw. i get 3,417s, built in 148ms