LumaTeam/Luma3DS #1487: Arm11 exception with Ridge Racer 3D

mallocup77, 2020-08-17:
[crash_dump.zip](https://github.com/LumaTeam/Luma3DS/files/5086452/crash_dump.zip)

**System model:**
New 3DX XL

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
v10.2

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Before)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Arm11 Exception Error when trying to play Ridge Racer 3D, both USA and EUR CIA versions.

Tried unchecking and checking the "Disable Arm11 exception handlers" option, as well as enable/disabling game patches.

The two dump files are from USA and EUR versions of the game.

**Steps to reproduce:**

1. Start the game.

2. Go through all the Single Player/Grand Prix process until the Events screen. When confirming, "an exception occurred" under Arm 11 (core 0).

**Dump file:**

mallocup77, 2020-08-18:
Update: this error doesn't not occur when using Luma v10.1.3.

AuroraWright, 2020-11-17:
Seems to not occur on latest

danylopez123, 2024-06-23:
Unfortunately, this problem it's happening again even with the latest v13.1.2
I do also get the same crash and i have the EUR version, this should be re-opened.

impeeza, 2024-06-23:
Hello there, I am using the file `0004000000033B00 Ridge Racer 3D (CTR-P-ARRP) (v0.0.0) (W).standard.cia` with SHA256:
362d175eb5175691b082231a91f5862217d1d6782273ac21e41ecd0396cfd575 *0004000000033B00 Ridge Racer 3D (CTR-P-ARRP) (v0.0.0) (W).standard.cia

Starting Luma 13.1.2 the game chrash before starting a race, you can star the game, create a new one and go to all the registration process, but when you accept the Race Event Select page, the game crashes.

Downgrading to Luma 10.3 the game runs fine.

impeeza, 2024-06-23:
[image]
[crash_dump_00000002[1].dmp](https://github.com/user-attachments/files/15945309/crash_dump_00000002.1.dmp)

danylopez123, 2024-10-10:
Guys, i've figured it out why this problem happens STILL with the new latest version of Luma:

Turns out the game will crash if you go to the Grand Prix Mode "if you choose the Clock or Clock+L2 option from New 3DS CPU"

PabloMK7, 2024-10-11:
Can reproduce, reopening.
@TuxSH

SonicBoone56, 2025-07-05:
Can confirm what @PabloMK7 says, running with increased clocks crashes the game when trying to select a race. However, you can increase clocks while in a race and if in a championship and selecting next race. You'll have to disable it before going in to the menu.