LumaTeam/Luma3DS #1494: 2ds xl bricks after installing

IMxKirby, 2020-08-30:
I am on 11.13.0-45U and just installed luma3ds 10.2 after using the safe mode exploit i got to the screen that says that luma3ds was installed press any button, I did and my 2ds turned off and wont turn back on or boot into safe mode again

luigiswashbuckle, 2020-08-31:
I am on 11.13.0-45U and just installed Luma3DSv 10.1.3 (i wasn't aware of 10.2) similar thing just happen to me, i did it on my "New 3ds XL" and it worked out just fine. I then, just tried it out on my "3ds XL" and it seems to have soft bricked it. it wont turn on, just a black screen, and the blue light turns on for about 2-3 seconds and turns off without turning on. this also happened after using the safe mode exploit, i got to the screen that says that luma3ds was installed press any button, after i pressed any buttton it turned off. any help would be grateful, thanks.

GerbilSoft, 2020-09-02:
The power light turning off after a few seconds usually means boot.firm wasn't found on the SD card. Make sure Luma3DS's boot.firm is present in the SD card's root directory.

dwain12435, 2020-09-21:
A solution has been given, so why is this issue still open?

AuroraWright, 2020-11-17:
Not a support forum

closed 2020-11-17 by AuroraWright