LumaTeam/Luma3DS #1496: Enable Game Patching [n3ds]

GamingInfinite, 2020-09-06:
I'm trying to randomize Pokemon Ultra Sun.  Followed all of the standard "put the right folders in the right patching areas" and all that.  I can't find anything wrong.  All the information I can give you is the path:

luma\titles\00040000001B5100\romfs

and that I've double, triple, and quadruple checked the way everything is laid out within the romfs folder.  I've got enable game patching on, and haven't messed with any other settings.

Ammako, 2020-09-07:
Ultra Sun is 1B50

dwain12435, 2020-09-21:
This issue has seemingly been solved, but I have experienced this same issue with all games that have update data installed.

AuroraWright, 2020-11-17:
Not a support forum

closed 2020-11-17 by AuroraWright