LumaTeam/Luma3DS #1498: Luma 3ds Chainloader not working?

HeroMines, 2020-09-09:
**System model:**

Old 3ds

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.2 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: (After )

Splash duration: (1 )

PIN lock: (8 )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( x)

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**
When I try to open the Luma 3ds Firm Chainloader on boot, it doesn't work, instead it just continues on to the 3ds menu. I have 4 files in it, so I don't think it has anything to do with Godmode9. Using the button shortcuts don't work, either. I have no idea what could be causing this, my start button works, so it's not that.

**Steps to reproduce:**

1. Hold start and try to access the Luma 3ds Firm Chainloader

Ammako, 2020-09-28:
boot.firm on SD card, or on CTRNAND?

HeroMines, 2020-11-14:
> boot.firm on SD card, or on CTRNAND?

Took so long to get a response, I forgot all about this lol. boot.firm on SD card I'm pretty sure.

AuroraWright, 2020-11-17:
Not a support forum

closed 2020-11-17 by AuroraWright