LumaTeam/Luma3DS #1499: An exception occurred. ARM11 - data abort - Translation-section

Luma208, 2020-09-14:
Processor : ARM11 (core 0)
Exception type : data abort
fault status : Translation - section
(I think)

I am new to not only the 3DS system but also to the Luma3DS stuff as well (my username is a coincidence). I followed the guide to install homebrew, and the latest version of the Luma3DS stuff (9.something I think). I bought a Japanese version of the Old 3DS, and followed the guide to install these things and change the region to the united states. I popped in a copy of "Yoshi's New Island," and it began to run. Despite this, when I selected a world to begin, the error screen would appear. Also, whenever I pop in a DS/DSi game and try to start it, the screen freezes black. Similarly, this occurs when I try to do DS download play for the DS/DSi system.

I can upload my dump somewhere, but I'm not sure how. I looked it up and tried alternative solutions, but I'm thinking I may have to downgrade to Luma3DS 8. I don't know how to do this, so I'm hoping to find a better solution before I do that.

AuroraWright, 2020-11-17:
Does it still happen on latest? and does the game region match the region of the installed firmware?

Luma208, 2020-11-17:
I should probably close this. It turns out that the game card itself was having issues as it ran other games just fine. Makes sense as I found the game card on the ground...Thanks for trying to help!

closed 2020-11-17 by Luma208

BojayB, 2021-01-12:
Im having this issue with an English cart of Majoras Mask using the Japan region. Ive been able to launch other English digital titles properly, and I’ve been able to launch English Smash Bros. so I’m a bit confused as to what is going on. At first I was unable to launch smash but it turns out I had the plugin loader turned on. I turned that on and it worked fine but I couldn’t launch Majoras Mask. I opened the Luma settings at launch and enabled game patching and I could launch Majoras Mask, but after I closed the game I couldn't open it again. This error showed prior to the game working and after it stopped. I’ve searched up a couple things but only got one answer and it hasn’t fixed anything.

BojayB, 2021-01-12:
Using the Luma title locale changer I changed MM to english and Japan region. I did the before and It didn't work but I tried again and its working now.  I’m going to figure out if it still works after a close or reboot.

AideGqre, 2024-04-16:
omg ty so much ! this seems to be the solution in 2024 too !!

Toox103, 2025-07-17:
> Using the Luma title locale changer I changed MM to english and Japan region. I did the before and It didn't work but I tried again and its working now. I’m going to figure out if it still works after a close or reboot.
> 
> Update: seems to work still. For anyone else having this issue later in time: Download the [Luma Locale Switcher](https://github.com/Possum/LumaLocaleSwitcher), select the title that you’re having translation issues with, and change the language and region to the carts original language and region. If you’re having this issue, it’s more than likely because you’ve got a foreign digital game or cart (best guess because my problem was because) or you’ve recently switched your O/N3DS’s locale/region to one that didn’t come with the console. GLHF!

even in 2025 it still work. ty bro