LumaTeam/Luma3DS #1503: ARM11 exception svcBreak in process menu

lateo96, 2020-09-24:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

I cannot check it anymore, but it's the latest American version.

**Entrypoint (How/what you're using to boot Luma3DS):**

I don't know how can I check it, but I followed the steps in the 3ds.hacks.guide page

**Luma3DS version:**

v10.2 stable

**Luma3DS configuration/options:**

`Screen brightness: 4(x)`

`Splash: Off(x)`

`Splash duration: 1(x) seconds`

`PIN lock: Off(x)`

`Enable loading external FIRMs and modules: ( )`

`Enable game patching: ( )`

`Show NAND or user string in System Settings: (x)`

`Show GBA boot screen in patched AGB_FIRM: ( )`

`Set developer UNITINFO: ( )`

`Disable Arm11 exception handlers: ( )`

---

**Explanation of the issue:**

The exception persists even without SD Card.

```
Processor: ARM11 (Core 0)
Exception type: prefetch abort (svcBreak)
Fault status: Debug event
Current process: menu (0004003000008F02)
```

[image]

I've tried a `CTRTransfer (D9)` using the script that comes with `GodMode9`. That was my best shoot. After completed I saw a "loading" dialog _"Preparing Home Menu"_ or something like that. After the loading finished I got the same error.

---

**Steps to reproduce:**

1. Turn on the 3DS

**Dump file:**
[crash_dump_00000000.tar.gz](https://github.com/LumaTeam/Luma3DS/files/5279672/crash_dump_00000000.tar.gz)

neoman1918, 2020-11-13:
I'm facing the same problem. Above message shows when I try to power on my O3DS XL. If I remove my SD card, the blue LED lights up, then goes off -- but I had already done CTRTransfer to NAND.

neoman1918, 2020-11-13:
Just managed to solve this problem by following the steps in the link below:

https://gbatemp.net/threads/black-screen-at-the-start-of-the-3ds.543980/page-3#post-8763629

closed 2020-11-17 by AuroraWright

Smoreslickgores, 2021-02-05:
Yo I have the same same problem so I went to fbi then went to titles then deleted the red things in titles cuz they were taking a lot of space on my 2ds then turned it off to take a break then the (Arm11) (core 0) (SvcBreak)

Masamune3210, 2022-05-06:
> Yo I have the same same problem so I went to fbi then went to titles then deleted the red things in titles cuz they were taking a lot of space on my 2ds then turned it off to take a break then the (Arm11) (core 0) (SvcBreak)

For the love of gods do not just go through the title list and delete titles you don't know about

Zatchhg, 2022-07-01:
I'm having the problem with core 1
[image]

Zatchhg, 2022-07-01:
[image]

1642128, 2025-01-10:
I learned that this error means there is missing code in the NAND (system files). So your best bet would be to use a NAND backup to reinstall the previous NAND using GodMode9 (https://www.cfwaifu.com/3ds-backup-restore/). If you can't do this (you don't have a backup) then you'll probably have to do some sort of factory reset.

Masamune3210, 2025-01-11:
There are loads of things that can cause this kind of crash, missing system files is only one cause.

lateus, 2025-01-11:
I'm the original poster of this issue. I just want to say that @Masamune3210 is totally right. This can have very different causes.

That said, I don't think that this issue was really fixed nor I think it should be opened again. There's not objective solution to this problem and, being realistic, people are not going to work on this anymore. I'm just thankful that Luma is as robust as it is, and this is not a common issue.

In my case, I tried a bunch of scripts one after another and eventually it got solved. That was more than four years ago, and in that time spam I've seen this error again under very different circumstances and they all got solved in very different ways, with most of them not being as severe as mine. GodMode9 usually comes with useful scripts, with a description of what they do and when you should use them. You can try those.