LumaTeam/Luma3DS #1506: Rosalina system crash playing Bowser's Inside Story

DarthVitrial, 2020-10-06:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**
FastBoot3DS

**Luma3DS version:**

 https://github.com/LumaTeam/Luma3DS/commit/ebeef7ab7f730ae35658b66ca97c5da9f663a17d

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: (off )

Splash duration: (1 )

PIN lock: ( off)

New 3DS CPU: (Clock+L2 )

--

Enable loading external FIRMs and modules: ( yes)

Enable game patching: (yes )

Show NAND or user string in System Settings: (tes )

Show GBA boot screen in patched AGB_FIRM: ( yes)

Set developer UNITINFO: ( no)

Disable Arm11 exception handlers: (no )

Enable Rosalina on SAFE_FIRM: (no )

--

**Explanation of the issue:**

While playing Mario and Luigi: Bowser's Inside Story + Bowser Jr's Journey, I my 3DS occasionally crashes with a system panic. Seems to either be when I open the start menu or after a battle ends - basically when transitioning from a subscreen to the overworld. Happens after an hour or so of playing, i would say?

I do have the  60fps cheat enabled, though I'm not aware of anyone else encountering such a crash with it: 
`D3000000 30000000
50000074 01000101
20000075 00000000
D0000000 00000000
520DA3AC 01000101
220DA3AD 00000000
D0000000 00000000
50000064 01000101
20000065 01000000
D0000000 00000000
50000044 01000101
20000045 00000000
D0000000 00000000
DD000000 00000100
50000074 01000001
20000075 00000001
D0000000 00000000
520DA3AC 01000001
220DA3AD 00000001
D0000000 00000000
50000064 01000001
20000065 01000001
D0000000 00000000
50000044 01000001
20000045 00000001
D0000000 00000000
D0000000 00000000`

Error message: Processor: Arm11, core 1.
Type: Data Abort.
Fault Status: Translation - Section
Current Process: Rosalina

Dump file:

https://mega.nz/file/090XHQJZ#hn26Iudc6I-BA7QmwHJGgy4Gdw-85lhpSq9fbaPduQg

DarthVitrial, 2020-10-08:
Along with Rosalina I get a number of crashes in Bowser's Inside Story where the current process is "Remillio (00040000001D1400)".
Should I post the crash dump for it too?

VenomRobbie, 2022-05-15:
Hey dude. It's been a few years but I'm getting this exact same issue in this exact same game. Did you ever find a fix?

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai