LumaTeam/Luma3DS #1507: Bravely Default randomly crashes

CaptainGotoSV2, 2020-10-07:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45E (SysNAND)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Frogminer

**Luma3DS version:**

v10.2 and the version before it

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Clock+L2) & (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Bravely Default (PAL version) randomly crashes throughout normal gameplay

**Steps to reproduce:**

There are no specific steps to reproduce the game crashing as this is random

**Dump file:**
[Bravely Default PAL Crash Dumps.zip](https://github.com/LumaTeam/Luma3DS/files/5338176/Bravely.Default.PAL.Crash.Dumps.zip)

Shadow322, 2020-10-07:
What method is this game ram from?

CaptainGotoSV2, 2020-10-07:
> What method is this game ram from?

Cartridge

xch25, 2020-11-19:
Interested here too

SamothJtM, 2020-11-26:
I think to have solved the problem by disabling my wifi. So that Bravely Default can't try to do the StreetPass thing. Can somebody confirm this is the problem? How could I disable StreetPass again completely for it again?

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai