LumaTeam/Luma3DS #1508: Obviously decompiled code in sysmodules/loader/source/loader.c

GabrielRavier, 2020-10-08:
https://github.com/LumaTeam/Luma3DS/blob/768e587b76a3b007e845e04cef66102b888facba/sysmodules/loader/source/loader.c#L29-L93

This code seems to obviously be direct output from IDA. Is this normal ?

profi200, 2020-10-08:
What do you mean "normal"? This code comes from [yifanlu](https://github.com/yifanlu/3ds_injector) originally. I guess he didn't bother cleaning it up.

GabrielRavier, 2020-10-09:
Well, I was wondering whether it was normal to have decompiled code in this kind of project. I've heard there could potentially be legal problems relating to use of reverse-engineered code, which could be problematic depending on the EULAs that Nintendo includes in their products. Ofc, this would only really apply if the code was written in the US (or another country with similar laws that forbid decompilation where the EULA does) and Nintendo had a clause in their EULA that forbids decompilation, and I am not really informed enough to know about either of those. 

I guess I'm not really that qualified to make an issue about this, considering my lack of awareness about the topic, but I've had a friend tell me that releasing Luma3DS under the GPL implies that there is no reverse-engineered code, which would then be wrong considering this code, which would be a serious issue.

urherenow, 2020-10-09:
You could have just Googles lzss. It's not Nintendo's IP.

GabrielRavier, 2020-10-09:
Where is the code decompiled from, though ? An implementation of an algorithm can still be copyrighted itself iirc, alike to a performance of classical music. I knew of LZSS as a compression algorithm, but even if the code is made to decompress LZSS data, that wouldn't change the fact that it's Nintendo's code. Ofc I'm not saying this has legal ramifications on the project : I have none of the necessary knowledge to make such a claim. But I'd like it to be confirmed that this is in fact ok (or a serious issue) without being told that "LZSS is not Nintendo's IP", which afaik is irrelevant, considering the code is still decompiled from Nintendo's implementation of LZSS.

profi200, 2020-10-09:
This code does reverse LZ77 decompression and yes, it got decompiled from the original Loader module. This being only a small piece of code i would say it falls under fair use. But i'm not a lawyer.

closed 2020-11-17 by TuxSH