LumaTeam/Luma3DS #1513: [FR] individual screens brightness adjust in Rosalina menu (N3DS only?)

eadmaster, 2020-10-24:
I've found these 2 implementations with the source code:
 - [tool by leojay_x](https://www.reddit.com/r/3dshacks/comments/7ehu0u/is_it_possible_to_have_separate_screen_brightness/)
 - [LCD Tweaker by MarcusD](https://gbatemp.net/threads/lcd-tweaker.461994/)

DullPointer, 2020-10-24:
It would work for both O3DS and N3DS. I tried a hacky adaption to allow individual screen adjustment [here](https://github.com/DullPointer/Luma3DS/blob/54c02cf561e1dd2a6d40265d802f9aeb27be2068/sysmodules/rosalina/source/menus.c#L167-L355). However, there seemed to be an issue with N3DS top screen coeffs so I used the O3DS ones [instead](https://github.com/DullPointer/Luma3DS/blob/54c02cf561e1dd2a6d40265d802f9aeb27be2068/sysmodules/rosalina/source/luminance.c#L117). That got it working approximately, but I'm not sure how to implement it properly.

eadmaster, 2020-10-25:
Even having an option to completely turn off one of the screens would be useful, because in some games the top/lower screen is barely used and it could help saving battery power.

I've tried the LCD Tweaker by MarcusD and it works while the app is running, but it resets as soon as i go back to the home menu.

eadmaster, 2020-10-27:
UPDATE: just found [this fork which has this feature implemented](https://github.com/DullPointer/Luma3DS/commits/master).

Can this be backported?

mooseyacky, 2020-12-01:
Please do this

lividhen, 2021-01-25:
I have a fork taking the only the brightness stuff from DullPointer's branch... I haven't had any issues with it on o3ds, but I don't have a new3s/2ds to test if it works on there... If you want I could pull out the individual screen stuff and submit a pull request. If it doesn't work on n3ds I am not sure how to fix it though...

CatsLover2006, 2021-02-08:
Just gonna mention that on the o2DS that this will have no affect (single backlight)

eadmaster, 2021-02-08:
> I have a fork taking the only the brightness stuff from DullPointer's branch... I haven't had any issues with it on o3ds, but I don't have a new3s/2ds to test if it works on there... If you want I could pull out the individual screen stuff and submit a pull request. If it doesn't work on n3ds I am not sure how to fix it though...
> https://github.com/lividhen/Luma3DS
> Here it is if you want to try it, but any issues with brightness will be the same as DullPointer's (as it is the same code) as of today.

I've already tried the DullPointer fork with my n3DS XL and it working fine, even if the UI for this feature is a bit ugly to use.

Btw for the models having no adjustable backlight simply hiding this option shoul be fine.