LumaTeam/Luma3DS #1514: Digital copies of Story of Seasons: Trio of Towns crash under certain circumstances if LayeredFS is used

ubergeek77, 2020-10-25:
**System model:** New 2DS XL

**SysNAND version:** 11.13.0-45U SysNAND

**Entrypoint:** Boot9Strap

**Luma3DS version:** v10.2 stable (this issue is not unique to this Luma3DS version, details below)

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

In digital copies of the game Story of Seasons: Trio of Towns (`000400000019f500`), if LayeredFS is active (even if no meaningful files are replaced or if a simple dummy file is added), the game will crash if the player has a "mushroom log" present on their farm and they enter the farm. This issue does not occur when using the cartridge version of the game.

I maintain a patch for this game that restores the character Hamtaro to the U.S. version of the game. I've had several reports over the years about a crash related to the mushroom log, but I could never find out why it happened or what caused it. It was clear LayeredFS was at least related to the patch somewhat, but I could never determine if this was a LayeredFS bug, or a problem with how I created the patch.

It was not until a report from today that I was able to determine the root cause - having LayeredFS active at all on digital copies causes the game to crash when trying to load a mushroom log, even if no game files are actually replaced by LayeredFS. This means the issue has been present in Luma3DS [for potentially 3+ years, at least since my first release of the patch](https://gbatemp.net/threads/release-story-of-seasons-trio-of-towns-hamtaro-restoration-patch.463794/) (reports of the mushroom log patch were reported to me shortly after that initial release).

This release of my patch predates the version 1.1 update for this game, and this type of crash was happening to users during that period. With that being said, the 1.1 update can most likely be ruled out as causing the crash.

A mushroom log takes many hours to unlock from a fresh game, and the only save file I have for testing this issue was given to me by a user playing on the 1.1 update of the game. It's also pretty baffling why this *specific* item is causing this crash, but it's completely reproducible, so something is going on here. In any case, while I am only able to reproduce on 1.1 due to only having a save on that version, I can basically guarantee this will also happen on an unpatched game.

As a final note, I should mention LayeredFS ***does*** work properly for this game, for the most part. Game files get replaced as you would expect. It's just that also having that mushroom log item on your farm leads to a crash.

**Steps to reproduce:**

1. Create the following blank file, just to cause LayeredFS to be active for this game*:

```sdmc:/luma/titles/000400000019f500/romfs/dummy.txt```

2. With LayeredFS enabled, load a digital copy of the game with a save file that has a mushroom log placed on the farm. Enter the farm, and notice the game has crashed.

_\*any file or combination of files present in this LayeredFS folder, whether or they exist in the original RomFS or not, will lead to a crash. The dummy text file was listed for reproduction purposes._

_

**Dump file:**

[arm11.zip](https://github.com/LumaTeam/Luma3DS/files/5434007/arm11.zip)

I'm not sure if the uploading of save files is allowed here, but I can upload a file that easily reproduces this if one is needed.

ubergeek77, 2024-01-31:
Hi there, I know this is an old issue, but this was tagged as "needs more research/info" around 2 years after originally reporting it.

Is there any info I can provide about this issue?

MareepSheepPeep, 2025-05-07:
Any chance this'll get revisited?
If you need more information, I'd be happy to help. The error screen looks like this:

**An error occurred (ErrDisp)
Error type: generic
Process ID: 38
Process name: BOKUJYO3
Process title ID: 000400000019f500
Address: 0x001120d4
Error code: 0xc8804470**