LumaTeam/Luma3DS #1519: Crash on latest FW update (11.14.0-46)

cgrs, 2020-11-17:
**System model:**

New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.14.0-46E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.2
**Luma3DS configuration/options:**

Default EmuNAND: (Off )

Screen brightness: (1 )

Splash: (Before )

Splash duration: (1s )

PIN lock: (Off )

New 3DS CPU: (Off )

--

Autoboot EmuNAND: (Off )

Use EmuNAND FIRM if booting with R: (Off )

Enable loading external FIRMs and modules: ( Off)

Enable game patching: (Off )

Show NAND or user string in System Settings: (On )

Show GBA boot screen in patched AGB_FIRM: ( Off)

Set developer UNITINFO: ( Off)

Disable Arm11 exception handlers: (Off )

Enable Rosalina on SAFE_FIRM: ( Off )

--

**Explanation of the issue:**

Afer updating to latest 3DS Firmware (11.14.0-46) and rebooting the console, an ARM11 exception appears.
Everytime the console boots Luma loads but the exception reappears.
It is possible to access payloads previously installed (I used this to boot into Godmode9 and boot NATIVE_FIRM. This made the console as if it was untouched, because after that I couldn't access Luma configuration screen and payloads).

**Steps to reproduce:**

1.Update to latest FW.

2. Reboot console.

**Dump file:**

(Unfortunately I didn't get a dump. I'll try to reproduce it again and upload the dump in this issue).

ThunderSpark91, 2020-11-17:
I got the same problem, thankx for posting this issue

AuroraWright, 2020-11-17:
It's being worked on

ThunderSpark91, 2020-11-17:
> 
> 
> It's being worked on

Thank you for the response but so far, my console is then completley bricked. Can anyone give me pointers to still acces my 3ds without the CFW? I have a crash dump but I can only open the 3ds tommorow to get the sd card out.

jcmejia, 2020-11-17:
Same here just gonna have to wait for the next luma update

nn9dev, 2020-11-17:
WEW LADS, I thought I was the only one. its crashing for me with AND without SD card in, New 2DS Xl as well. 
crash_dump_0000000.dmp is here https://files.catbox.moe/wn531k.dmp

AuroraWright, 2020-11-17:
Fixed in https://github.com/LumaTeam/Luma3DS/commit/94b5113b17a54d8595a0d2195f06b27bfd73ee79

closed 2020-11-17 by AuroraWright

minexew, 2020-12-17:
Out of pure curiosity, what does commit 94b5113b17a54d8595a0d2195f06b27bfd73ee79 actually change in terms of the generated code? To me, the source code change would seem like a no-op!

profi200, 2020-12-17:
Notice how [svcKernelSetState()](https://github.com/devkitPro/libctru/blob/master/libctru/include/3ds/svc.h#L1187) takes a variable number of args? If you pass args with the wrong types that's not gonna work very nicely when the kernel expects something else.