LumaTeam/Luma3DS #1522: Minecraft 3ds download play stops working

mooseyacky, 2020-11-17:
I added Luma3ds and the download play for Minecraft stopped working

TrickPlayzMC, 2020-11-18:
Minecraft for the 3ds does not have download play

mooseyacky, 2020-11-18:
I meant local multiplayer

AuroraWright, 2020-12-01:
Could you contact me on Discord at Aurora Wright#9538 ? This is a bit difficult to test because two New 3DSes are required

mooseyacky, 2020-12-01:
I tried to contact you but I can only do that from discord if I am friended so I sent a friend request. You can delete me right after if you want. 
[image]

dwain12435, 2020-12-04:
I did the test on latest USA Minecraft version (1.9) with Luma3DS v10.2.1-3gxldr on both consoles and multiplayer worked perfectly fine.

mooseyacky, 2020-12-07:
Hmmmmm

closed 2021-03-11 by mooseyacky