LumaTeam/Luma3DS #1523: 3ds crash after official update

marcs7, 2020-11-18:
After update my n3ds to 11.14.0-46E when booting it crashes, is there any way to resolve this problem ?

Shadow322, 2020-11-18:
> After update my n3ds to 11.14.0-46E when booting it crashes, is there any way to resolve this problem ?

U this blind?

marcs7, 2020-11-18:
> > After update my n3ds to 11.14.0-46E when booting it crashes, is there any way to resolve this problem ?
> 
> U this blind?

why ?
it has installed luma3ds and i update it, i'm asking if is a solution to fix this

regulad, 2020-11-18:
update to newest luma 10.2.1 
just read the changelogs man

marcs7, 2020-11-18:
> update to newest luma 10.2.1
> just read the changelogs man

yes i did it, i found the solution in discord homebrew server, update it manually and it worked

CatsLover2006, 2020-11-18:
could you close this issue?

closed 2020-11-18 by marcs7

Cota-UABC, 2021-01-10:
I can see this community is really nice.