LumaTeam/Luma3DS #1527: This is it. My sister is gonna kill me.

usnsjhgiufdifbyfd9ughfd, 2020-11-19:
I've had a 3DS for about 6 years now, and the newest update bricked it. Even trying to insert the new boot.3dsx and .firm versions did not work.

Fuck you Nintendo.

hifinerd, 2020-11-19:
Try using a fresh boot.firm. The newest Luma3ds is designed with 11.14 in mind.

xch25, 2020-11-19:
> Try using a fresh boot.firm. The newest Luma3ds is designed with 11.14 in mind.

How can I do this?

Shadow322, 2020-11-19:
> > Try using a fresh boot.firm. The newest Luma3ds is designed with 11.14 in mind.
> 
> How can I do this?

By replacing it

closed 2020-11-21 by TuxSH

Dimensional, 2020-11-22:
I think it's safe to assume this guy is probably using a n3DS and doesn't know how to remove the mSD card from it to connect it to his PC and from there replace the file.

SirEnder125, 2020-12-11:
I think that you may need to manually install the .firm
with gm9 > [0:] SD Card > .firm file > FIRM options > Install Firm > Both

SirEnder125, 2020-12-12:
Oh, that is if your mind works enough to
boot into luma chainloader and install gm9.
(Lol jk, who doesn't)

GlaZedBelmont, 2020-12-13:
That's definitely not the best thing to do, also this issue is a month old.