LumaTeam/Luma3DS #1528: Load alternate CIAs/Home Menus on boot

hifinerd, 2020-11-19:
It would be cool to see an option to run a specific app when booting Luma3ds, so you can load an alternate Home Menu alongside the regular one.

mooseyacky, 2020-12-01:
There is an app for this

hifinerd, 2020-12-03:
@mooseyacky What is it called?

SirEnder125, 2020-12-12:
Sounds cool, what IS it called?

urherenow, 2020-12-12:
Perhaps he's alluding to Sunshell...

mooseyacky, 2020-12-12:
3dsbank. to get it here is the link: https://gbatemp.net/threads/release-3dsbank-nintendo-3ds-folder-selector.516996/

NightYoshi370, 2020-12-14:
@mooseyacky That's unrelated. He wants to boot into another application on boot, not have separate banks for the home menu.

mooseyacky, 2020-12-14:
Read the title, "Load Alternate Home Menus on Boot"

NightYoshi370, 2020-12-14:
And you should read the actual post: "It would be cool to see an option to run a specific app when booting Luma3ds"

Also, he probably means something like the Homebrew Launcher or TWiLight Menu++ when he says "Alternative Home Menus"

mooseyacky, 2020-12-15:
@NightYoshi370 Then they can add new startup boot combos to load apps during boot like Homebrew Launcher, 3DSBANK, or TWiLight Menu++ or a custom app so that they don't have to wait.

suppai, 2022-06-05:
Duplicate: #1582

closed 2022-06-05 by suppai