LumaTeam/Luma3DS #1529: Error 2ds sin tarjeta sd

Flekyyo, 2020-11-19:
Hola buenas, yo quiero reportar un error que me sucedió al actualizar la 2ds.
Primero le tengo instalado la ultima versión del Luma, pues actualice mi 2ds a la última versión oficial, le coloque el parche nuevo de luma para q funcionara y todo de maravilla, pero si se apaga la consola y la enciendes sin tener puesta la tarjeta sd da este error:
[image]
Si se presiona A dos veces sin la sd puesta se apaga o presionando cualquier botón.
Al poner de nuevo la sd y encenderla, presionar dos veces A, todo vuelve a funcionar como si no uviese pasado nada, pero esto con la anterior versión de Luma no pasaba, quitavas la sd y la 2ds funcionaba correctamente, con su programa original.
Espero que mi aporte le sirva para intentar reparar esta incidencia, un saludo.

Alvaro1307, 2020-11-19:
Eso es debido a que para que la consola con CFW de Luma funcione, tiene que estar el archivo boot.bin de la última versión de Luma en la raíz de la tarjeta SD para que encienda la consola. Debido a la última actualización de 3DS, Luma no era compatible, por tanto como dice usted, ha descargado el último parche de Luma y colocado en su tarjeta pero sigue apareciendo el error si la enciende sin Tarjeta SD, eso es por que para arrancar sin tarjeta también necesita el archivo boot.bin, en este caso el archivo va dentro de la CTRNAND. En YouTube puede encontrar vídeos de como actualizar Luma en la CRTNAND, es muy simple. Espero que le ayude

Shadow322, 2020-11-19:
English please

Flekyyo, 2020-11-20:
Hola de nuevo, antes de nada daros las gracias x la ayuda prestada.
E visto videos de como actualizar Luma cambiando los dos archivos, y si funciona todo bien.
También he probado a restablecer la nand.
Pero sin éxito, sige todo igual, sin tarjeta sd, la 2ds es un bonito posa baso.
Pero bueno no me voy a quebrar mas la cabeza, lo dejaremos como esta q x lo menos funciona bien.
Un saludo a todos/as y gracias x la ayuda.

closed 2020-12-01 by AuroraWright