LumaTeam/Luma3DS #1530: Errdisp

milotusauto, 2020-11-19:
System model:

[New 3DS XL]

SysNAND version (+emu/redNAND version if applicable):

[11.14.0-46U SysNAND]

Entrypoint (How/what you're using to boot Luma3DS):

[Boot9Strap]

Luma3DS version:

[v10.2.1 stable]

Luma3DS configuration/options:

Default EmuNAND: ( )

Screen brightness: 4(x)

Splash: Off(x)

Splash duration: 1(x)

PIN lock: Off(x)

New 3DS CPU: Off(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

Explanation of the issue:
When playing MHGen, sometimes I get Errdisp on loading screen.

**Steps to reproduce:**

1. Play MHGen

2. Go to new area to trigger loading screen
[image]

Dump file:
Did not provide dump file

Oreo639, 2020-11-19:
The save data for the game is probably corrupted.
https://zoogie.github.io/error/?d900458b

milotusauto, 2020-11-20:
> The save data for the game is probably corrupted.

That explains why I kept getting error "The archive could not be opened. 0xC8804497" when i try to save data on JKSM. Is there any way to fix this? It is a cartridge game.

Oreo639, 2020-11-20:
MHGen doesn't put it's save data in the "save data" section, it puts it in the extra data section. You would probably need to delete the extra data for the game.

milotusauto, 2020-11-20:
>You would probably need to delete the extra data for the game.

Deleting the extra data worked for a while then the problem returned. I realized the SD card was the one causing the corruption, so I changed it and now it works perfectly. Thanks for the help!

milotusauto, 2020-11-20:
It came back...

closed 2020-11-21 by TuxSH

milotusauto, 2020-11-25:
Turns out the cartridge was defective.