LumaTeam/Luma3DS #1531: Problem with Smash

Zetonium, 2020-11-20:
Model: New 3DS XL 
Region: US
Problem: This might not be very descriptive but every time I launch smash I get this error 
Processor: Arm11 (core 0) 
Exception Type: data abort
Fault Status: Permission - Section
Current process: ctrapp (00040000000EDF00

[image]

Dump I cannot link because I am on mobile but ask me in the comments and I'll post it.

Keep in mind this happened a few weeks ago randomly and also I don't care about my save because I got this game on October 18th so I don't really care if a method deletes my save

AuroraWright, 2020-12-01:
Does the folder luma\titles\0004000000EDF00 exist on SD and if so what's inside it?

Zetonium, 2020-12-03:
code.ips

dwain12435, 2020-12-04:
Try re-downloading or re-dumping Super Smash Bros. If it still doesn't work, then make sure your SD Card isn't corrupt via h2testw (ask @TurdPooCharger on GBAtemp for a tutorial).

AuroraWright, 2020-12-05:
@Zetonium Delete the file and try again

Zetonium, 2020-12-05:
THANKS SO MUCH IT WORKED AND MY SAVE IS STILL INTACT now please close the thread

closed 2020-12-05 by Zetonium