LumaTeam/Luma3DS #1532: Arm11 exception and errdisp

abejonejo, 2020-11-20:
**System model:** 

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.14.0-46E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

v10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
After updating to 11.14 and to Luma 10.2.1, I installed Yokai Watch Blasters White Dog and when I launched the game, I got an ARM11 exception. Then I decided to download the new update for the game (v2.4)
[arm11exc_and_errdisp.zip](https://github.com/LumaTeam/Luma3DS/files/5572978/arm11exc_and_errdisp.zip)
 even it wouldn't boot. After that, tried to launch the game again and ARM11 exception disappeared but got an ErrDisp instead. I'm attaching both dumps.

I tested my SD card with f3 and it was ok.

**Steps to reproduce:**

1. Open Yokai Watch Blasters White Dog

2. Get ARM11 exception if the game is not updated, or get ErrDisp if game update is installed

**Dump file:**

splapoon, 2020-11-20:
You most likely got those from an application crash, try reinstalling it.

TuxSH, 2020-11-21:
Corrupted game installation

closed 2020-11-21 by TuxSH