LumaTeam/Luma3DS #1535: New 3DS XL crashes on startup, most likely cause is a system update

DartRuffian, 2020-11-25:
**System model:**

New 3DS (XL)

**SysNAND version (+emu/redNAND version if applicable):**

Cannot check, DS crashes on boot

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

Luma3DS v10.1.3 (No idea beyond that)

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: ( Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
3DS crashes on launch, I'm able to open the Luma settings but if I just let it boot up it crashes. I tried using the backup of the "gm9" folder from when first homebrewing the 3DS but it still crashed (included dump files for both the backup and normally). I believe this is because my brother installed a system update on his DS, if it helps, I have my homebrewed new 3DS XL (so the same model) w/out any system updates since I homebrewed it.

**Steps to reproduce:**

1.

2.

**Dump file:**
[Dump_file.zip](https://github.com/LumaTeam/Luma3DS/files/5594725/Dump_file.zip)

tomjschwanke, 2020-11-25:
This was fixed in the latest release:

1. Download the latest version from [https://github.com/LumaTeam/Luma3DS/releases](https://github.com/LumaTeam/Luma3DS/releases) (the zip file)
2. Take out the SD Card from your 3DS and connect it to the PC
3. Copy the two files from the zip file to the root of the SD Card (not into any folder)
4. Put the SD Card back into the 3DS, boot up and enjoy.

DartRuffian, 2020-11-25:
> 
> 
> This was fixed in the latest release:
> 
>     1. Download the latest version from https://github.com/LumaTeam/Luma3DS/releases (the zip file)
> 
>     2. Take out the SD Card from your 3DS and connect it to the PC
> 
>     3. Copy the two files from the zip file to the root of the SD Card (not into any folder)
> 
>     4. Put the SD Card back into the 3DS, boot up and enjoy.
> 
>     5. Go into the Homebrew menu and select the Luma3DS updater to update the installation on CTRNAND (this version is booted if no SD Card is inserted)

Alright, I downloaded the latest version and it booted up fine, I was tempted to try it but figured it would be best not to in case that would cause a different problem. Thank you for the fix!

closed 2020-11-25 by DartRuffian

rays, 2020-12-01:
Tried these steps and I had to boot via GodMode9 and boot firmware manually but the end result was the same. Crash on boot

Thisguy-me, 2021-08-16:
I have the Same problem