LumaTeam/Luma3DS #1537: An exception occurred ARM11 (core 1) / Exception type: undefined instruction / I need help!

davidgd72k, 2020-11-27:
Hi, I have a problem with booting my 3DS, I hope somebody can fix this problem.
I have a **Old 3DS (EUR)** with **Luma3DS v10.0.1** (now I can't know the SysNAND version, sorry). The console using Boot9Strap.

About the origin of the error, I don't know very much what could have happened, I only know that after playing a little bit of a game, the next day the error started to occurred the error.

So, every time I power on the console, I get this on the screen:

**An exception ocurred**

Processor: ARM11 (core 1)
Exception type: undefined instruction
Current process: pm (0000000000000000)

Attach a picture with more info (sorry for low light level):

[image]

More interesting info:

**Luma3DS configuration/options:**
Screen brightness: 4(X)
Splash: Off(X)
Splash duration: 1(X)
PIN lock: Off(X)

Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )

**Dump file:**
[crash_dump_file.zip](https://github.com/LumaTeam/Luma3DS/files/5609379/crash_dump_file.zip)

I try with other ways but nothing :(

By beforehand, thanks.

davidgd72k, 2020-11-27:
P.D.: the error screen appears with the SD card inserted and also without the SD card inserted.
And I don't have a Nintendo 3DS cartridge inserted in the console.

profi200, 2020-11-27:
Update to the latest Luma version manually (replace boot.firm on the root of the SD card). The newest update broke older versions of Luma.

MicrolemurDev, 2020-11-27:
I also has this issue and I can confirm that profi200's response does work on the New 2DS XL. Kudos to him/her!

davidgd72k, 2020-11-28:
I finally solved the problem by doing what say profi200; replace old Luma3DS files for updated Luma3DS files on the root of my SD Card.
For now my 3DS works well, nothing bad I've seen.

Thanks for all.

TechManPrieto, 2020-11-28:
My friend has experienced this issue and replaced the files, and it seems to work fine now. Well, I am going to archive this as a point of reference in the future.

ryanzb, 2020-11-30:
Replacing the Luma files fixed three New 2DXLs so I copied the files to an Old 3DS as well, for good measure!

TechManPrieto, 2020-11-30:
> Replacing the Luma files fixed three New 2DXLs so I copied the files to an Old 3DS as well, for good measure!

I replaced them on my old 3DS XL and my friend did as well on his old 3DS. We got them at launch... geez that was long ago.

closed 2020-12-01 by AuroraWright

rays, 2020-12-01:
Updated the files and rebooted to the same error. If I ask GodMode9 to boot the firmware it loads. Is there a step I'm missing to getting the device to boot

Shadow322, 2020-12-04:
> Well I was told to delete all of my files from my SD card before I saw this, but now I don't know if I will ever be able to update luma and fix the DS

ROTFLMAO why the hell told u to do that?

ryanzb, 2020-12-04:
> > Well I was told to delete all of my files from my SD card before I saw this, but now I don't know if I will ever be able to update luma and fix the DS
> 
> ROTFLMAO why the hell told u to do that?

I hope no one tells him about the Windows\System32 folder

CringyCraze, 2020-12-04:
I can still access my Godmode menu is there anything I can do with that?

Neo564, 2020-12-15:
Hi guys, I have the same problem, but switching the update files didn't do anything.
I formatted my SD Card too, thinking it would help.
Does someone know how to fix this ?

I'm greatful for your ideas :)

whiskyfly, 2020-12-15:
I have a new 2dsxl and I have the same problem, but switching the update files didn't do anything.
I'm greatful for ideas too!

AbhorRend, 2020-12-16:
This happened to me after an update on my 3DS. I tried updating Luma, but not too sure if what im doing is right, as my screen never changes

Oreo639, 2020-12-17:
You need to download the zip from https://github.com/LumaTeam/Luma3DS/releases/latest then copy the boot.firm to the root of your sd card.

AbhorRend, 2020-12-17:
> You need to download the zip from https://github.com/LumaTeam/Luma3DS/releases/latest then copy the boot.firm to the root of your sd card. If there is already a boot.firm, delete it.

Thank you so much. I did this before but must've used the wrong files, cuz it worked with those. Appreciate the help

whiskyfly, 2020-12-17:
For everyone who is unsuccessful, test with a smaller micro sd. That way it worked.

chimpisbadwithcomputers, 2020-12-29:
> Update to the latest Luma version manually (replace boot.firm on the root of the SD card). The newest update broke older versions of Luma.

My name is completely correct how the heck do I do that?

itismartin, 2021-01-06:
Also have the same problem as Neo564, but I haven't formatted my SD card yet.

tcslk, 2021-01-24:
> > Update to the latest Luma version manually (replace boot.firm on the root of the SD card). The newest update broke older versions of Luma.
> 
> My name is completely correct how the heck do I do that?

@chimpisbadwithcomputers 

Ok so try downloading this zip: 

https://github.com/LumaTeam/Luma3DS/releases/download/v10.2.1/Luma3DSv10.2.1.zip

And copy the two files (boot.firm and boot.3dsx) from the zip to the root of your SD card (not in any folders in the SD card) and put the SD card back in the 3ds and try that.

kylemart, 2021-02-02:
Wanted to share my own findings. In addition to updating the boot.firm and boot.3dsx like other have suggested, I also needed to update GodMode9 to get the [START]+[POWER] combo working again.

LursoPolar, 2021-04-09:
I did the same with my 3ds, but for some reason it's not updating the luma, it keeps on 10.0.1 and not the actual. even though i did the same process, downloading the last lumavers. and copy and paste on the root of the sd... what can i do?

ITZM4Z, 2021-04-18:
@profi200  at first i was a bit lost but finally it worked! thank you so much!

mcradcliffe2490, 2021-04-18:
I replaced both files with the new Luma v10.2.1 ones and no dice. System still crashes. Working on an original 3ds. Anyone got any advice?

mcradcliffe2490, 2021-04-18:
> I replaced both files with the new Luma v10.2.1 ones and no dice. System still crashes. Working on an original 3ds. Anyone got any advice?

Nevermind I'm a goodball. The SD card wasn't formatted properly. The fix in this comment section works.

EnderBossGT, 2021-07-07:
Im using the latest firm but i still get that error when i launch dragon quest stories. Btw im on a NEW 3ds XL.

mcradcliffe2490, 2021-07-07:
@EnderBossGT Specifically only dragon quest stories? Have you tried uninstalling the rom after updating luma and trying again?

bianca199864, 2021-08-23:
I have a Nintendo 2DS that has an exception occurred and I don't know how to get off my screen I need help

d8958101, 2021-09-23:
thx a lot, problem solved.

SepticRedhead, 2021-10-07:
Mine says Arm11 core1 exception type undefined instruc process loader 
[image]
0000000000000000

thhreaper72806, 2021-12-02:
hey i need help i have this problem to and i cant get it to work i tried updating the file but it wont work pls help i have the new 3d
[image]
[image]
s

mcradcliffe2490, 2021-12-02:
@thhreaper72806 did you use the guide specific to the new 3ds when installing luma?

thhreaper72806, 2021-12-02:
yes

thhreaper72806, 2021-12-02:
ive had my 3ds for a like a year homebrewed

mcradcliffe2490, 2021-12-02:
@thhreaper72806 did you recently do a system update on your 3ds?

thhreaper72806, 2021-12-03:
yes but nothing was wrong after the update i updated the firmware and was fine

mcradcliffe2490, 2021-12-03:
@thhreaper72806 Did you update luma before you did the system update?

thhreaper72806, 2021-12-03:
after and it was working btw if you have discord that would make this much easier

mcradcliffe2490, 2021-12-03:
I'm not an expert but I've been using homebrew for a while, you can hit my disc and we can see if the issue can be sorted out BigPapiKite#2836

thhreaper72806, 2021-12-03:
ok it is sent accept the friend request and we can vc thx for staying with me this far

bryan24G, 2022-01-12:
thank you this was an easy fix.

WilburNoot69, 2022-05-08:
> Update to the latest Luma version manually (replace boot.firm on the root of the SD card). The newest update broke older versions of Luma.

it never worked for me?? i replaced the boot.firm file and the boot.3dsx file. it hasn't worked? I've got the exact same issue :((

Liliaimilk, 2022-08-25:
> Update to the latest Luma version manually (replace boot.firm on the root of the SD card). The newest update broke older versions of Luma.

wow,I did it😊

getapples, 2023-03-18:
I'm a little late but I have replaced the boot.firm 3 times and nothing has happened. My 2ds is still stuck on the Arm11 error screen. Anyone know what I should do?

Shokamii, 2023-08-25:
> Update to the latest Luma version manually (replace boot.firm on the root of the SD card). The newest update broke older versions of Luma.

Godspeed to this unknown hero

Dangisuckathomebrew, 2024-02-12:
I’m struggling with this also, though my console still functions. I was too excited when hacking it and didn’t eject my sd card even once and after I put it into the console it wouldn’t read it. After putting it back onto my laptop it said I had to format it. Whenever I try to open the homebrew launcher by opening download play I get this error. Is this possible to fix? Thanks

Zushikina-kun, 2024-10-23:
FIX for ARM11 Core 1 Error on 3DS (Loading External Storage Firms and Modules)

If you've tried everything suggested here and you're still getting the ARM11 Core 1 error, here's something else you can try that worked for me:

Back Up and Delete Your luma/titles Folder:

Backup the luma/titles folder on your SD card (just in case).
Delete the luma/titles folder from the SD card.
This folder isn't critical to the system itself, so deleting it won't break anything important. It typically contains custom patches, cheats, mods, or region-free patches for specific games or the system. If you're using any of those, you can easily re-download and set them up again later.

Don't worry—your save data will remain safe.

Why This Works:

The error is often caused by a corrupted external storage firm or module in the luma/titles folder (or possibly in luma/modules, though the latter may not exist on all setups). I don’t know the exact cause of the corruption, but removing this folder resolved the issue for me.
My Experience:

Initially, unchecking "Enable loading external storage firms and modules" in the Luma3DS settings made my 3DS work again, but I couldn’t access Pretendo servers with that setting off.
After researching online and brainstorming with ChatGPT, I finally found that deleting the luma/titles folder fixed the problem without needing to disable external storage firms and modules.

Masamune3210, 2024-10-24:
......you know what the problem with using LLM's to do your homework for you is? About half of that is wrong. 

You don't need to use GPT to follow instructions, one of the first troubleshooting steps for almost anything modding related is to back up the sd card, wipe it, and extract fresh copies of needed files

Masamune3210, 2024-10-28:
No. I don't think I will. 

Still doesn't stop half of that being some variant of incorrect. Turns out a random chat model being abused into doing something it was never designed to do doesn't know shite. Who would have thought.

Nemris, 2024-10-30:
Masamune is right on the money. It's not like LLMs existing means we can turn off our brains.

@TuxSH Perhaps this issue should be locked at this point? Looks like it got derailed quite a bit over the years.

Alkaloon72, 2024-11-26:
Having the "and exception occurred" error on my old 3ds, anyone have ideas? 
[image]

Alkaloon72, 2024-11-26:
It seems updating Luma WITHOUT updating the "luma" folder or the "boot strap" folder fixed this issue