LumaTeam/Luma3DS #1540: Arm11 exception crash on Level 8-5 of Chicken Wiggle (Atooi Collection)

RupeeClock, 2020-11-29:
**System model:**

New 3DS XL, New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45E SysNAND (N3DS)
11.14.0-46E SysNAND (N3DS XL)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( n/a )

Screen brightness: ( 4 )

Splash: ( Off )

Splash duration: ( 1 )

PIN lock: ( Off )

New 3DS CPU: (Off / Clock+L2  )

--

Enable loading external FIRMs and modules: ( On or Off )

Enable game patching: ( On or Off )

Show NAND or user string in System Settings: ( On )

Show GBA boot screen in patched AGB_FIRM: ( Off )

Set developer UNITINFO: ( Off )

Disable Arm11 exception handlers: ( Off )

Enable Rosalina on SAFE_FIRM: ( Off )

**Explanation of the issue:**

When playing Chicken Wiggle as part of the Atooi Collection, an exception will occur when attempting to begin level 8-5 of the Play mode.
This crash is consistent on both the physical game card and on .CIA dump installations on two separate N3DS series systems running Luma v10.2.1
Different Luma3DS configurations concerning N3DS CPU, external firm loading or game patching did not change this.

**Processor:** Arm11 (core0)
**Exception Type:** data abort
**Fault status:** Permission - Page
**Current process:** ATOOICOL (00040000001D9300)

**Steps to reproduce:**

1. Launch Atooi Collection title, begin Chicken Wiggle game

2. Enter Play mode, go to level 8-5, select level and click Play

**Dump file:**
[crash_dump_00000000 (v11.13.0-45E N3DS).zip](https://github.com/LumaTeam/Luma3DS/files/5612272/crash_dump_00000000.v11.13.0-45E.N3DS.zip)
[crash_dump_00000006 (v11.14.0-46E N3DS XL).zip](https://github.com/LumaTeam/Luma3DS/files/5612273/crash_dump_00000006.v11.14.0-46E.N3DS.XL.zip)

**Save file:**
[Atooi Collection Save.zip](https://github.com/LumaTeam/Luma3DS/files/5612271/Atooi.Collection.Save.zip)
A save file of Chicken Wiggle that has been played up to level 8-5 is provided, if it is of any use.

RupeeClock, 2020-11-29:
Arm11 exception crash will also occur on level 8-6 if you have a save game that has the final level unlocked/completed.
I was able to play the last two levels using Citra Canary build 1887, and transfer that save across to my N3DS XL.

AuroraWright, 2020-12-02:
The game was tested on stock by a friend and it also crashes there. My friend also found out that this crash was even reported by someone on the Facebook page of the publisher. Tl;dr not our issue

closed 2020-12-02 by AuroraWright

RupeeClock, 2020-12-02:
Disappointing to learn that it's a fault with the released game, given the nature of its release, but thank you for checking this.