LumaTeam/Luma3DS #1541: Selecting a version from the BootNTR

Rubiks-20, 2020-12-01:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
SysNAND 11.14.0-46U

**Entrypoint (How/what you're using to boot Luma3DS):**
boot9strap

**Luma3DS version:**
v10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: (Before payloads)

Splash duration: (3 seconds )

PIN lock: (Off )

New 3DS CPU: (Clock )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

When launching BootNTRSelector and selecting a BootNTR version from the menu, it crashes! I tried clicking on the buttons fot the other versions, but it didn't work.

**Steps to reproduce:**

1.

2.

**Dump file:**
[crash__dump_00000029.dmp.zip](https://github.com/LumaTeam/Luma3DS/files/5620179/crash__dump_00000029.dmp.zip)

Shadow322, 2020-12-01:
Not a Luma3DS bug that's all BootNTR

DavidMB60, 2020-12-01:
> Not a Luma3DS bug that's all BootNTR

Apparently not, if you see [BootNTR 2.13.4 release](https://github.com/Nanquitas/BootNTR/releases/tag/v2.13.4) there's a bug in [Luma memory mnapping svc](https://github.com/LumaTeam/Luma3DS/blob/master/sysmodules/rosalina/include/csvc.h#L70-L79). This is all from the creator of BootNTR. So maybe it's an issue of both programs.

TuxSH, 2020-12-07:
It's a kernel bug, not my fault

closed 2020-12-07 by TuxSH