LumaTeam/Luma3DS #1543: Problem with Nintendo contract

Jose123456789007654, 2020-12-02:
I can´t accept Nintendo´s contract and I can´t play online or use spotpass or stretpass.

Jack06WS, 2020-12-04:
Raising issues are like this are ridiculous. Nobody can help you if you spend 5 seconds on your issue request. Let us know lots of information such as your Luma3ds version, device model, firmware version. Provide images or videos.

Nobody will put effort in to help you if you can't put effort in for 5 minutes to type a helpful issue.

dwain12435, 2020-12-04:
Check to see if this is a Luma3DS issue by doing the OFW safety test:

1. Navigate to the correct folder for your device using GodMode9 (accessible by holding down START before and during boot and releasing it when the Luma3DS chainloader screen appears and then pressing A).

Old 3DS(XL)/2DS - [1:] SYSNAND CTRNAND/title/00040138/00000002/content
New 3DS(XL)/2DS XL - [1:] SYSNAND CTRNAND/title/00040138/20000002/content

2. Select the APP file.
3. Select NCCH image options.
4. Select Mount image to drive.
5. In the window that appears after pressing A, open the exefs folder.
6. Select the FIRM file.
7. Select FIRM image options
8. Select Boot FIRM
9. When prompted, choose to decrypt and make a copy.
10. You should boot into the regular 3DS Home Menu. If you do, then try updating your console. If it works, then the failed update is a Luma3DS issue. If it doesn't work, then the failed update is not a Luma3DS issue and it should be posted somewhere else.

Jose123456789007654, 2020-12-06:
I did all of that, but when I get into the 3ds home page a few games that I had dowloaded dissapeard and when I updated my console it appeard it was updated.

dwain12435, 2020-12-07:
Strange. Did you try restarting your console?

Jose123456789007654, 2020-12-07:
@dwain12435 If you mean by turning off and turning on the console, I did it and it didnt change anything.

AuroraWright, 2020-12-07:
@dwain12435 what kind of advice is that? Please refrain from posting it. By following that procedure, if the update includes a NATIVE_FIRM update, you undo the b9s install (which is what happened to @Jose123456789007654 here probably). @Jose123456789007654 please install b9s again.

AuroraWright, 2020-12-07:
About the issue at hand (which is completely unrelated to updating the console), that happens when the game is out of region. You need to use this homebrew app to set the EULA for all regions https://github.com/noirscape/dses/releases . Closing here as it's not a support forum

closed 2020-12-07 by AuroraWright

Jose123456789007654, 2020-12-07:
Hi @AuroraWright I might sound a little stupid but how do I install b9s?

AuroraWright, 2020-12-07:
Follow the procedures to hack your 3ds again as if you'd never done it

Shadow322, 2020-12-07:
> Hi @AuroraWright I might sound a little stupid but how do I install b9s?

https://3ds.hacks.guide/

Jose123456789007654, 2020-12-07:
Thanks all!💞