LumaTeam/Luma3DS #1544: My 3DS can not update and won't allow me to do anything until I do. HELP!

SirEnder125, 2020-12-02:
[Old 2DS]

I am pretty sure that this is only happening on my system. Not at all sure why.
I can not do anything that requires internet and when I try to update, it says

> An error occurred while trying
> to update your system.
>
> The system will now restart.

The progress bar moves about a quarter of an inch, then gives me that error and
reboots.
    I'm sure you really don't care, and you are probably going to redirect me to some
website where help is offered, I will not go to random websites you give me links to,
so if you can offer help, do so. If not, please don't reply, I WILL insult you.

Thanks.

dwain12435, 2020-12-04:
1. Open Luma3DS config by holding down SELECT during boot and disable developer unit mode if it is enabled.
2. Get a CTRTransfer image of the latest version from somewhere OR investigate whether you have a faulty/damaged wireless chip.

SirEnder125, 2020-12-05:
What is CTRTransfer?
Also, how would I go about seeing if I
> have a faulty/damaged wireless chip?

SirEnder125, 2020-12-05:
Sorry to rush you but if I don't figure it out soon I might just reset my 2ds. I have ways of
getting my friends back and I can backup my games through .cia files.

Also, is there a way to backup my NAND and restore it after I reset my ds so that I can come see my old
stuff every now and then?

dwain12435, 2020-12-06:
If you want to try resetting your console, then you shouldn't do it the
normal way. Here is the procedure you should follow:
1. Open GodMode9 by holding down START before and during your console's
boot and releasing it as soon as you see the Luma3DS chainloader screen and
then pressing A.
2. Navigate to [1:] SYSNAND CTRNAND
3. Rename the "data" folder to "dataBACKUP".
4. Boot your console and go through initial setup.
5. Try to update. If it works, then go back to GodMode9, delete "data", and
rename "dataBACKUP" to "data".
6. If it doesn't work, then download 3DS version 11.14.0-46U through NUS.
Also, 3DS version 11.13.0-46U (which is what you said you already
downloaded through NUS) doesn't exist. Maybe a typo?

SirEnder125, 2020-12-06:
Will this overwrite my SD card?

dwain12435, 2020-12-07:
No.

SirEnder125, 2020-12-07:
Ok, if I backup my NAND first can I restore it to get everything back just in case?

dwain12435, 2020-12-07:
Yes, just use a script in GodMode9. You should already have a script that
can do this.

SirEnder125, 2020-12-07:
I'm confused, I think, will my current NNID be gone, or no?

closed 2020-12-07 by TuxSH

SirEnder125, 2020-12-07:
The method you told me didn't work, I renamed data to dataBACKUP and my console
wouldn't boot.