LumaTeam/Luma3DS #1545: help olds 3ds wont work

ki-krpto, 2020-12-02:
everytime i boot up 3ds arm11 (core 1) processor type pm comes in
started this morning
and it says it gives me the dump but it does not
anyone know how to fix?

Shazer51, 2020-12-04:
im having the same issue

dwain12435, 2020-12-04:
You have to copy the "boot.firm" file from the Luma3DS archive to the root of the SD Card and overwrite the already existing one.

ki-krpto, 2020-12-05:
that did not work but i will take a picture 
here you go
[image]
[image]
sorry if its blurry

TuxSH, 2020-12-07:
Drag and drop boot.firm onto the root of the SD card, replacing the existing file.

When you press Select at boot, does it say it's been booted from SD card (yellow text bottom-left corner bottom screen)

closed 2020-12-07 by TuxSH