LumaTeam/Luma3DS #1546: Crash when playing cartridge game/demo

MizuhoChan, 2020-12-03:
**System model:**
[New 3DS XL]

**SysNAND version:**
[11.13.0-45E SysNAND]

**Entrypoint:**
[Boot9Strap]

**Luma3DS version:**
[v10.2.1 stable]

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Clock+L2, also tried Off)

Enable loading external FIRMs and modules: (Off)

Enable game patching: (Tried on and off)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (On/Off)

Set developer UNITINFO: (On/Off)

Disable Arm11 exception handlers: (On/Off)

Enable Rosalina on SAFE_FIRM: (Off)

**Explanation of the issue:**

When I attempt to load the Zelda: Majora's Mask Not for Resale Demo cartridge, the game will crash.

An exception occured
Processor: Arm11 (core 0)
Exception type: data abort
Fault status: Translation - Section
Current process: Joker (000400000015AD00)

**Steps to reproduce:**

1. Play the cartridge
[crash_dump_00000002.zip](https://github.com/LumaTeam/Luma3DS/files/5638923/crash_dump_00000002.zip)

**Dump file:**
Attached

MizuhoChan, 2020-12-03:
I will add that I tried it on a japanese O3DS that had been hacked just to play games region free, this also did the same thing.

pspalmirasara, 2020-12-23:
I was having a similar issue with Majora's Mask JPN (same expection, process, fault status... dump here: [crash_dump_00000005.zip](https://github.com/LumaTeam/Luma3DS/files/5736585/crash_dump_00000005.zip)). 
I think it was a region issue, I was able to fix it with LumaLocaleSwitcher.

MizuhoChan, 2020-12-24:
Oh wow, that worked.

I guess I can confirm that it doesn't load on EUR or JPN.

Thanks for the help pspalmirasara 👍

closed 2020-12-24 by MizuhoChan