LumaTeam/Luma3DS #1548: Black screen when starting some software on old 3ds

Atkael, 2020-12-05:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.14.0-46E

**Entrypoint:**
Homebrew? Couldn't fully figure out this one.

**Luma3DS version:**
v10.2.1

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (off)

Splash duration: (1)

PIN lock: (off)

--

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**

Some titles result in a black screen upon launching. Examples are: Super Smash Bros, Flip note 3D, Kirby battle royale.

**Steps to reproduce:**

1. Install a game (SSB or Kirby battle royale, for example)

2. Launch it

3. Some result in a black screen and never load

takeshineale128, 2020-12-05:
you have to wait about 20 secs for the game to load

Atkael, 2020-12-05:
It doesn't work on mine. I launched it and several minutes later still only a black screen.

nickolay-smyk, 2021-02-20:
Same issue here. Some games just get stuck on black screen, no new crash dumps are generated. At the same time the titles that don't freeze on load - run perfectly fine. Tried waiting about 10 mins on the black screen - no results. The only way to proceed is to force-shutdown the console by pressing power button for about 5-8 sec.

Console: Old 3DS LL

Luma ver: 10.2.1-c058e3f (latest version by now)

Luma Settings:
Screen brightness: (3)
Splash: (before)
Splash duration: (1)
PIN lock: (off)
--
Enable loading external FIRMs and modules: (x)
Enable game patching: (x)
Show NAND or user string in System Settings: (x)
Show GBA boot screen in patched AGB_FIRM: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )

List of titles that I was able to reproduce the issue with:
- Detective Pickachu
- Luigi's Mansion
- Captain Toad Treasure Tracker

Extra info that I can add is that if I turn volume all the way up when launching these titles - I can hear a sound like speakers are "overloaded", so it seems that there occurs some kind of voltage spike like when you momentarily shut down speakers from power source. From this I assume that there occurs some kind of soft-shutdown of the console at this point.

It's very frustrating. I was able to find many threads with description of similar issues (with no solution unfortunately), so the problem is definitely global. Would really appreciate if someone would shed some light here.

Mewthree3, 2021-04-01:
I often have the same issue (with a region changed tp Japanese N2DSXL. It's happened with Zelds: Twilight Princess, Smash 3DS (happened today actually), and Rhythm Heaven (although it somehow resolved itself).
One of the things that helps at times is to use Luma Locale Changer. It worked for Rhythm Heaven. The problem is pretty global annoyingly enough. I'll try waiting 20 minutes+ to see if it helps.

Spiralka, 2022-01-07:
I have same issue, It's happened with Mario Maker for me. I can see 3ds's launchscreen and then - black screen

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai