LumaTeam/Luma3DS #1551: [Rosalina Feature Request] Add option to go into sleep mode from Rosalina menu

proteekrosul, 2020-12-10:
I see the 'Power off' and 'Reboot' options in the Rosalina menu but is it possible to add an option to manually put the system to sleep mode?

SirEnder125, 2020-12-11:
Not to be rude, but, I do not see a point for the developers to add a function that
can be called upon in an VERY easy way by pushing a switch. Oh, wait,
I do not own a 3ds. A 2ds. Do the not have sleep mode? Of course they
do, it is just enabled by folding the 3ds with those (flimsy) hinges.

proteekrosul, 2020-12-12:
> 
> 
> Not to be rude, but, I do not see a point for the developers to add a function that
> can be called upon in an VERY easy way by pushing a switch. Oh, wait,
> I do not own a 3ds. A 2ds. Do the not have sleep mode? Of course they
> do, it is just enabled by folding the 3ds with those (flimsy) hinges.

I realize what you are saying but unfortunately my 3ds shuts down when folded, it can be kept on only in the right-angle kind of position. And I have seen this happen to a few other people with 3ds too.

lividhen, 2021-01-31:
You could get a small magnet and place it just to the right and upish from the X button.

proteekrosul, 2021-02-08:
> 
> 
> You could get a small magnet and place it just to the right and upish from the X button.

Yeah but then it shuts down when lifting the lid

closed 2021-02-17 by proteekrosul