LumaTeam/Luma3DS #1552: Emunand crash with wifi

Makein, 2020-12-10:
**System model:**

[New 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.14.0-46E SysNAND, 11.2.0-35E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[v10.2.1, try 8.1 , 9.0 and 9.1 too]

**Luma3DS configuration/options:**

Default EmuNAND: 1(x )

Screen brightness: 4(x )

Splash: Off(x )

Splash duration: 1(x )

PIN lock:Off(x )

New 3DS CPU: Off(x )

--

Autoboot EmuNAND: (x )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

After booting in emunand with wifi ON after few seconds it's crash, if wifi disabled all fine.
When I enable it it's crash after few seconds. In settings menu all fine.
This thing started after I update luma3ds+b9s+SysNAND to 11.14.0-46E.
Set there account, changed mii, download pokemon shuffle.
Then go back to emunand and try enable wifi, crash (it's was working before)
ps: sysnand works fine...
[crash_dump_00000001.zip](https://github.com/LumaTeam/Luma3DS/files/5676244/crash_dump_00000001.zip)

**Steps to reproduce:**

1. Boot emunand

2. Enable wifi

**Dump file:**

Makein, 2020-12-10:
Oh, and I check  SD Root/Nintendo 3DS/(32 Character ID)/(32 Character ID)/extdata/00000000/ and there no 0000008f folder
if it matters...

AuroraWright, 2021-06-14:
Can't reproduce, N3DS XL, Japanese Emunand.

closed 2021-06-14 by AuroraWright