LumaTeam/Luma3DS #1554: [Rosalina Feature Request] Add Option to Return to HomeScreen or Homebrew Channel

MarioKartFan, 2020-12-11:
Sometimes homebrew will hang and the system will no longer respond to the HomeKey. Nevertheless, the Rosalina menu can still be pulled up and used to reset the system. It would be amazing if we could use it to “Quit” the current program and return to either the HomeScreen or the HomeBrew Channel!

mooseyacky, 2020-12-12:
I agree, that would be very very useful!

T0biasCZe, 2020-12-14:
also add "reset" button which will reset the game homebrew would be great =) 
(like the physical reset button on wii, gc, n64, snes and nes but digital)

MarioKartFan, 2020-12-15:
That would be incredible.

DullPointer, 2021-01-02:
I did a little bit of experimentation trying to terminate hanging apps (ModMoon and "Portal 3DS") using [https://libctru.devkitpro.org/ns_8h.html#aa451e546d4aa3cb2692767f0a20c63c3](NS_TerminateProcessTID), but I just found that it would either throw an arm11 exception or cause a pop-up that would force you to reboot the console anyway :/.

INFINITE-PIKA, 2021-03-02:
> I did a little bit of experimentation trying to terminate hanging apps (ModMoon and "Portal 3DS") using [...]

Try to terminate a non-hb title and see what happens, maybe?

suppai, 2022-06-05:
Duplicate of #1516.

closed 2022-06-05 by suppai