LumaTeam/Luma3DS #1555: Brick

sadbocks, 2020-12-12:
Im on a new 3ds on the latest update that have been on CFW for quite a bit of time. I was messing around with the settings of my various CFW applications but issues aroused when I opened Luma. It gave me an error message telling asking me if I would like to update and after multiple attempts, it worked. I changed the bottom setting (I really dont remember what it did) And when I pressed start my console turned off and turned back on with a black screen. Im able to turn on my system and the light will stay blue but the screen is black. I have a feeling this is only a problem with me and Ive started asking and searching for help on other websites. If you can patch this and make sure it doesnt happen to others please do.

minexew, 2020-12-17:
Are you able to boot back into Luma3DS options by holding Start?

datslammedcl9, 2020-12-27:
Someone please update this thread? I have the same issue my girlfriend was messing around with my 2ds XL and she deleted something by accident and I’m stuck with a black screen when starting and a blue led, I can still access godmode 9

saint-etienne, 2021-01-19:
restore your sysnand backup using godmode

INFINITE-PIKA, 2021-01-30:
BOOT INTO SAFE MODE! That's saved me before!
L+R+A+dpad up on boot

lividhen, 2021-01-31:
On the discord, if you have this issue, they recommend turning it on, and leaving it alone for 15-20 minutes, and if that doesn't fix it, clearing your exdata (don't remember where that is, you'd have to ask them...)

MicrolemurDev, 2021-02-16:
Try re-installing Luma3DS .firm files. I had a black screen after updating and it worked for me!

**Disclaimer:** It may not be the problem but it never hurts to try!

AuroraWright, 2021-06-14:
Seems possibly the same as https://github.com/LumaTeam/Luma3DS/issues/1099 (check my reply there)

closed 2022-06-05 by suppai