LumaTeam/Luma3DS #1556: Pokémon UltraMoon Crash Errdisp 0xC960454C address: 0x00346688

Umbreon-Dlc, 2020-12-12:
Hello, I had to replace my 32gb memory with an original 8gb hc and when I wanted to install Pokemon Ultra Moon Fbi again, it gave me the error 0xc940454c, I tried to install it by Qr and when starting it gave me the same error code, that is, this:

-------------------------------------

Error type:       generic

Process ID:       49
Process name:     momiji
Process title ID: 0x00040000001b5100

Address:          0x00346688
Error code:       0xc960454c

-------------------------------------

I already remove the DSP1, tricks, change the luma and gm9 and homebrew and this still happens, sometimes it opens but passing the necrozma intro it returns to the error, in the eshop it tells me to repair it but it does not leave because the ticket is illegitimate , I already tried to remove the update I think the same thing will happen if I remove the save again.

splapoon, 2020-12-15:
try reinstalling.

AuroraWright, 2021-06-14:
Try redownloading and using a different SD card (in case your SD card is fake)

closed 2021-06-14 by AuroraWright