LumaTeam/Luma3DS #1558: Luma pin security breach

wuniversales, 2020-12-16:
## Description
When you turn on the console a text is displayed that says "insert 4-digit pin".
This is really ugly, the interface should be more visual.
There is also a security breach...

## I propose to make the following changes:

1 - Don't show how many digits the password has. This for safety, of course.
2 - Delete "START to shutdown" (For that there is the shutdown button, it is not really necessary to press start to shutdown.)
3 - Show an image of a padlock on the touch screen.
4 - Show the same image on the upper screen, but with a smaller size.
5 - Place the asterisks of the " ***** " pin below the upper image as the user enters it.

## Visual Example
**Upper display**

Show padlock image centered.
Show pin (asterisks) with "Placeholder" text "Insert your pin code", centered.

----
**Lower screen**

padlock image centered

urherenow, 2020-12-17:
I like this idea, if it's even possible to show pictures this early in the boot process, but I missed what you're saying the "security breach" is...

profi200, 2020-12-17:
I don't think you know what a security breach is... And that's ignoring the fact the pin is not meant for security in the first place. It's meant to prevent children from messing with the settings. The pin can be bypassed deleting Luma's config.

wuniversales, 2020-12-17:
In my case I have luma installed in the nand.
Also his thing is not to show the characters that the key has.
If the system is installed on the nand, no one can find out the pin key.

profi200, 2020-12-17:
Except ntrboot/any exploit that gives you access to the NAND.

TuxSH, 2020-12-17:
Deleting the pin+config also works. I agree with not showing the number of characters but that's low priority.

The main design point is to prevent the key combo from being recovered.

> In my case I have luma installed in the nand.

This shouldn't be your primary environment, luma-on-nand has some limitations and only exists as a fallback.

AuroraWright, 2020-12-17:
Even if you have luma-on-nand you can bypass it by putting in a SD card with a boot.firm on it anyway heh

wuniversales, 2020-12-18:
It depends, in my case I do not allow the automatic startup of the firm through the sd, as a priority I use the luma of the nand, maybe there are other methods to enter that I do not know ...
Likewise, the length of the pin should not be shown for obvious reasons.

Dimensional, 2020-12-19:
So you've configured the B9S bootloader to prioritize Nand over SD? Not bad, but much like how B9S was discovered and exploited, anyone with a B9S flashed flashcart and magnet can bypass the settings and boot from their own stuff, like GM9 or something else, and reverse that setting. So keep all the magnets away from your 2/3DS. :D

urherenow, 2020-12-19:
I custom-built my ntr card to boot the b9s installer as ntrboot.firm instead of boot.firm, so I can do everything in one step (no swapping out the sd card in-between to replace safeb9sinstaller with Luma).

So yea... pretty darn easy.

Nuck-TH, 2020-12-19:
Guys, do you get that pin code is access limiting feature, and it was never intended as robust security solution?

urherenow, 2020-12-19:
Since pretty much all main contributors and the owner have chimed in already, I'm pretty sure they know better than you, what the intent was.

AuroraWright, 2020-12-20:
@wuniversales if you're protecting against someone who has physical access to your 3DS and can even insert their own SD card, imo it's rather pointless to recompile B9S when they can use a ntrboot cart to bypass it.

wuniversales, 2020-12-29:
Well, I don't know if there will be any "Solid" solution for this, I just suggest the changes.
I think that at least, it would be good not to say the number of characters that the key has.
Then the graphical interface is a matter of aesthetics.

closed 2021-06-14 by AuroraWright