LumaTeam/Luma3DS #1559: Every homebrew Title will freeze on start after playing New Super Mario Bros 2 Gold Editon

Galel, 2020-12-20:
**System model:**

New 2DS XL & New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.14.0-46J SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.3

**Luma3DS version:**

10.2.1

**Luma3DS configuration/options:**

Screen brightness: (4 )

Splash: ( off)

Splash duration: (1 )

PIN lock: (off )

New 3DS CPU: (off )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( y)

Show NAND or user string in System Settings: (y )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Homebrew Titles will freeze on start after playing New Super Mario Bros 2 Gold Editon,(And the JP region normal edition ).
Homebrew that I’ve tried: FBI, checkpoint, mgba, temperPCE, virtualNES, picodrive, Snes9X(All installed from a cia). 
Strangely enough, if I run some other 3DS title before launching a Homebrew all would be fine, no freeze at not.

**Steps to reproduce:**

1. Run “New Super Mario Bros 2 Gold Editon”

2. Run Any Homebrew app just following that would cause the homebrew freeze on launch.

**Dump file:**
No err log found. Hope someone could look into it, love luma so much!

saint-etienne, 2021-01-19:
Confirmed. You have to open then close an official nintendo app first like system settings, activity log, etc.. to fix the bug.

VauZed, 2021-03-06:
I know that this is a bad thing but I am happy, that this is not some kind of wierd problem wich only I am experiencing

TuxSH, 2021-06-18:
Can reproduce on Standard Edition NSMB2, this seems to be a libctru/citro3d-related GPU crash issue, possibly related to something the game does wrong.

Not a Luma issue, but are any other games doing that?

Searinox, 2023-09-28:
I am not only able to reproduce this issue 100% with New Super Mario Bros. but have discovered that Castlevania Mirror of Fate causes the same issue: launch the game, close it, then launch the Homebrew Menu and the console will hang with a black screen. Console is EUR New 3DS XL with Luma3DS 13.0.2 and latest firmware. If you launch some other game after the offending titles, homebrew will launch as normal.

mariomadproductions, 2024-11-05:
I seem to be having this issue with normal NSMB2 on EUR New 3DS console.

TuxSH, 2025-04-08:
Fixed in latest libctru, though this requires recompilation of homebrew applications if they are used directly. Using the updated hbmenu in recent Luma3DS updates should prevent this issue from occurring.

closed 2025-04-08 by TuxSH