LumaTeam/Luma3DS #1561: Keeps crashing on startup

fgre396, 2020-12-29:
**System model:**
Old 3ds
[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
I cant check
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
boot.firm i think
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v10.2.1
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
if i try to turn it on it just crashes immediately, but it still allows me to boot up godmode9 and the update screen.
 
Also if i disable arm11 exeption handles then it would just bring me to a black screen and if i press the power button it brings me to that screen telling you about sleepmode (as it should) but if i tap the home button it freezes on the part where it says "home menu"

**Steps to reproduce:**

1.

2.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/5751909/crash_dump_00000000.zip)

Andres20141684, 2021-01-01:
I have this same issue. Tried to delete the extdata as other issues suggested, but no luck. 

Particularly, Luma is 8.0v. Hacked with bootstrap and haven't updated the nintendo since forever. Maybe 11.04. New nintendo 3DS.

Now I don't know if I have to update Luma and bootstrap. I want to know if thats the reason why deleting extdata is not working.

urherenow, 2021-01-01:
If you have to ask if you need to update, then you're being rude for posting here. Posting bugs about a old versions is a useless waste of time because the bug could have already been fixed. You should be using Luma3DS 10.2.1, if not building the latest commits yourself.

Andrew8102, 2021-01-02:
Maybe you should update your luma3ds firmware and delete your delete your home menu extdata as #824 did. I update my 3dsll and it crashed as you did, and I finally repaired it by using above approaches. Hope it could be a lettle help

TrashyOne, 2021-01-22:
Try turning off **Enable loading external FIRMs and modules**. I have been looking how to fix an issue with that setting crashing my 3DS without any luck.
I want to use the Twlight widescreen patch, and without any sysmodules in the folder at all, it seems to crash either way, so the function might be broken on some (if not all) systems.

Darkwolf64-max, 2021-03-18:
my issues started after installing a custom theme, it did the same things fgre mentioned, but with the latest version.

k4yb4, 2021-03-23:
> my issues started after installing a custom theme, it did the same things fgre mentioned, but with the latest version.

i'm having the same problem. Pls someone help.

closed 2022-06-05 by suppai