LumaTeam/Luma3DS #1563: [Enhancement] Power Saving Sleep

froggestspirit, 2021-01-03:
Would we be able to have a toggle to disable some power consumption items? i'm mainly thinking the processes running while in sleep mode for street pass, pedometer, etc., and to maybe lower the clock speed while in sleep? I'm hoping this could boost battery life while in sleep mode, since some of these items are seldomly used by people nowadays. I've seen there was a request open for disabling power to the cart slot, would this help in general if no game is in the slot?

profi200, 2021-01-03:
Clock can't go any lower than 268 MHz which is the max. on o3DS and the lowest on N3DS. It already goes into pretty much the deepest sleep the hardware supports. The best you can hope for is blocking processes running under sleep. But not sure if they run in the first place if WiFi is disabled.

TuxSH, 2021-01-04:
The thing that's happening with SpotPass/StreetPass/eShop download is that they just wake up the console at regular intervals and go back to sleep. Most sysmodules and the rest of the processes are outright suspended during sleep mode. The battery impact should be negligible, even more so if comms are disabled.

During sleep mode, all processors are in low-power wait-for-interrupt state (basically the only interrupt that can wake the 3ds CPUs are wake events), the GPU and screens are off, VRAM is powered down and FCRAM is in self-refresh mode.

closed 2021-01-04 by TuxSH