LumaTeam/Luma3DS #1564: the legend of zelda : majora's mask sd:/luma/dumps/arm11/crash_dump_00000004.dmp

NheByS, 2021-01-06:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

NheByS, 2021-01-06:
new 3ds
im new at github sorry for my all mistakes

Juancarlo577, 2021-01-23:
That form is used to diagnose what problems and how to reproduce them this has little to no information on how to solve it also you should attach the dump file as well as what model of 3ds and system version via the form to get a clearer way to get help

INFINITE-PIKA, 2021-01-29:
You're probably fine unless if it happens on lots of tries at booting the game (without cheats ofc, cheats do those a lot)

closed 2022-06-05 by suppai